morgendagen/riden-dongle #43: RD6030

Nuqvxz, 2024-10-05:
You can add RD6030 (id=60301)?

closed 2024-10-14 by morgendagen

morgendagen, 2024-10-14:
Hi,

I have created a new release with untested support for the RD6030 (I do not have access to one).

The main issue is that I do not know the correct V, I and P multipliers to use, so I assume they are the same as for most other models (i.e. 100).

Please check that you get the correct values when reading V, I and P. I'd be grateful for feedback so that I can fix any issues and remove the "untested" designation.

Nuqvxz, 2024-10-16:
Everything works great with RD6030.

hrandib, 2024-12-02:
Hi @morgendagen,
I use a firmware without RD6030 support on RD6030 and it looks like something is wrong with P value, others are fine.
Output examples:
27.460 V / 23.520 A / 645.850 W
27.480 V / 24.020 A / 4.700 W
27.500 V / 28.030 A / 115.460 W

I'd say there is an overflow somewhere

hrandib, 2024-12-04:
It means that RD6030 uses more than 1 holding register to keep this value

hrandib, 2024-12-05:
It should affect all models from RD6012

PR #46 (2024-10-14): Added (untested) support for RD6030.
Fixes #43