morgendagen/riden-dongle #47: Power output value is truncated

hrandib, 2024-12-06:
Power output value uses 2 holding registers, all the models that can yield more than 655W are affected.
See https://github.com/morgendagen/riden-dongle/issues/43

morgendagen, 2024-12-07:
Hi @hrandib,

Thank you for looking in to this. Are you able to provide a pull request? I'd be happy to merge it :)

hrandib, 2024-12-07:
It'll take some time

hrandib, 2024-12-08:
Didn't manage to compile the project out of the box. ESP libraries in PIO have been updated:
-    WiFiClient newClient = tcpServer.accept();
+    WiFiClient newClient = tcpServer.available();

-    uint32_t uart_baudrate = std::strtoull(uart_baudrate_string.c_str(), nullptr, 10);
+    uint32_t uart_baudrate = strtoull(uart_baudrate_string.c_str(), nullptr, 10);

commit c9634287 (2024-12-15): Fix the power output truncation. Resolves #47 (#48)
closed 2024-12-15 by morgendagen