morgendagen/riden-dongle #49: Interfacing with Python (Pyvisa) requires some work

hb020, 2025-02-20:
My main automation platform is around pyvisa (plus some custom code for prologix). I guess that goes for a lot of people.
It would be great if this code would be supported by pyvisa as well.

2 issues 'out of the box':

* pyvisa will not discover the device by itself. What it supports is:
  * VXI-11 portmap (port 111, on UDP or TCP)
  * MDNS discovery of "hislip" (not lxi, and not scpi-raw)
* communication with the device requires setting of `instrument.read_termination = '\n'`

The latter is not an issue, but I'd really like to get discovery working. Am still looking if I can find a bypass via Python. And if that doesn't work, I am willing to see if I can get either portmap or hislip working.

Will keep this ticket updated. If in the meantime you find a solution, that would be much appreciated.

Minimal python test code for discovery:

```python
import pyvisa
rm = pyvisa.ResourceManager()
resources = rm.list_resources(query="?*")
print(resources)
```

hb020, 2025-02-23:
In the meantime, I found some other problems with the socket interface: it can easily be locked up when multiple commands are sent in short order. As I'm working on a VXI-11 server, I can also look at that issue in my fork.

morgendagen, 2025-02-25:
Hi. Some sort of VXI-11 og HiSLIP support would be fantastic. I took a cursory look at it, but for my immediate requirements it  was not required.

hb020, 2025-02-25:
I have the insides of a hislip server, it even handles discovery, but would require a lot more testing and some more code, as it takes 2 simultaneous sockets, and the state engine and the handshaking are rather involved. The coming week I won't have time for it, but I could try to work on that after that. There is some space free in the ROM, so why not. But I'd prefer to take that on in a second phase. 

Also, many more devices are VXI-11 compatible than hislip compatible. hislip is complicated, and it shows in the support. Of my 5 more serious ethernet capable devices, all support VXI-11, 3 support both scpi-raw and scpi-telnet, and only 1 supports hislip.

closed 2025-03-01 by hb020