morgendagen/riden-dongle #55: Add support for new type of wifi dongle

bind-w-exit, 2025-03-01:
[image]
ESP8684-wroom-04c
PCB Have another pins conected

[image]
[image]
[image]

hb020, 2025-03-01:
ESP8684 = ESP32-C2. Is the successor of ESP8266 that is end of life now. But platformio does not support it (yet).
See https://github.com/platformio/platform-espressif32/issues/1151

It might be possible to get this working.

tinyboxxx, 2025-05-10:
my new RD6030Max is using same ESP8684 board, need this firmware too, I can test it for you if you can build a bin?

hb020, 2025-05-10:
The ESP8684 (or ESP32-C2) is not officially supported yet, and while there are almost daily commits to jason2866's repo, it is still very much bleeding edge. As a result, while it would be possible to produce a binary, the probability of it working would be very low. And debugging by email or message on github is going to be close to impossible: there are way too many things that can break or are broken, and would require specific tooling and test setups. I really prefer testing with hardware on hand first.

Having said that, I have tried compiling a binary, but even that doesn't work (yet). I'll need to dig in deeper.

hb020, 2025-05-12:
Just ordered one of the new Wifi modules. Will arrive by the end of the month, and will be able to do some debugging then. In the meantime, please know that the old modules are still widely available (in the new form factor, but still ESP12F, just search for "rd6006 wifi module").

hb020, 2025-05-31:
Update: As mentioned in the linked PR, I have found that Riden has prevented the ESP8684 from being reprogrammed. Both new dongles I have are encrypted. So: retrofit with ESP8266 is the only easy solution. 

One could also place a new ESP8684 on it, but then a new set of hurdle arrives:
* platformio does not yet support the esp32-c2 chip, you need access to beta versions of platformio
* it is an esp32, so the firmware needs some adaptation and testing.

I have prepared all of that, but it is not stable, so please do the retrofit for now.

hb020, 2025-06-02:
Shall we close this?

morgendagen, 2025-06-02:
> The PR is merged. Shall we close this?

Yes, thank you for reminding me. And thank you for looking in to this issue. I had just received a couple of the new dongles, but alas, I may have have ended up with some pretty useless paperweights...