morgendagen/riden-dongle #56: beeper is inverted?

hb020, 2025-03-17:
Slightly redacted info:
```python
>>> print(inst.query('*IDN?'))
Riden,RD6012,0000XXXX,1.36
>>> print(inst.query("SYST:BEEP:STATE?"))
1

>>> inst.write("SYST:BEEP:STATE 0")
>>> print(inst.query("SYST:BEEP:STATE?"))
1

>>> inst.write("SYST:BEEP:STATE 1")
>>> print(inst.query("SYST:BEEP:STATE?"))
0
```
After that, the web interface says "Buzzer enabled | No"

Something wrong here, but totally minor issue.

morgendagen, 2025-03-18:
Thank you for the bug report. I will have a look at it later this week.

hb020, 2025-03-18:
no problem, I can work on that. I'll might add some commands that could be helpful.

morgendagen, 2025-03-23:
Nice catch. Just verifier it on my RD6006 running UniSoft V1.41.1k.

```
$ lxi scpi -a RD6006-00001104.local -r "SYSTEM:BEEPER:STATE 0"
$ lxi scpi -a RD6006-00001104.local -r "SYSTEM:BEEPER:STATE?"
1
$ lxi scpi -a RD6006-00001104.local -r "SYSTEM:BEEPER:STATE 1"
$ lxi scpi -a RD6006-00001104.local -r "SYSTEM:BEEPER:STATE?"
0
```

The write command is working as expected (0 => muted, 1 => unmuted) as you noted.

I would lean towards removing support for the `SYSTEM:BEEPER:STATE?` command since it returns incorrect information.

hb020, 2025-03-23:
Or just document it. The problem is also on the modbus interface. So it is everywhere.

morgendagen, 2025-03-26:
Turns out RD6006 firmware V1.23 (non-UniSoft) does not exhibit this bug, so I did as you suggested, and documented the issue.

PR #57 (2025-03-26): Modbus register 69 (buzzer enabled) may return an incorrect value
Fixes #56
closed 2025-03-26 by morgendagen