morgendagen/riden-dongle #60: Proposition: simple interactive web interface

hb020, 2025-06-02:
What I often find myself doing is opening the web page 'details' and pressing refresh frequently. That could be done better.

My proposition: simplified UI.
  * auto refreshing output V,I, output mode (CV/CC), batt mode and output state (on/off)
  * allow setting V and I, and output on/off
  * cherry on the cake: allow executing any SCPI command, and seeing the return

Better would be an auto refreshing screenshot with buttons, but I haven't found a way to do that via the modbus interface.

I can make some mockups.

morgendagen, 2025-06-02:
That would be awesome :-)

I had contemplated doing something like this a while ago, but never got around to it. Maybe even including graphs, which could be done using e.g. [D3](https://d3js.org) in the browser to offload the dongle.

Ideally the dongle could provide a continuous stream of measurements over a WebSocket connection to the browser. This will require some rework of the web server, possibly using ESPAsyncWebServer. I did this in another project, although with an ESP32. It might be that the ESP8266 is just not powerful enough...

A simpler approach would be for the browser to poll the dongle for measurements at regular intervals. That would provide a good starting point before looking into using a WebSocket.

morgendagen, 2025-06-04:
Cool! Graphing capabilities can be added later anyway. I have used D3 fairly extensively at work a few years ago, so I will probably look at adding graphs when you are ready. D3 can be loaded from a CDN, and html, javascript and css for a simple graph can be done in 5KB.

hb020, 2025-06-05:
The basic web page is ready. 

[image]

Am looking into graphing now. 

Tell me if you want a PR on the basic page first or if you want to wait until the graph arrives.

morgendagen, 2025-06-05:
If you want to do the graphs, I won't hold you back, and will wait for the PR.

hb020, 2025-06-06:
see PR #61

Gummy35, 2025-06-15:
Seems like we had the same idea at the exact same time !
I had a bunch of esp32 and started the work on my fork 2 weeks ago, and added elegantOTA, WebSerial (to handle safemode, reboot, scpi commands, etc), and a customizable webui using websockets.
We also had exactly the same idea of querying only a subset of the PSU variables.
As your changes are merged into the base repo, I will try to refactor my code to be closer to yours when possible,